craftbeerpi/craftbeerpi #232: Install.sh errors related to WiringPi

gbucey, 2020-04-29:
I am encountering the following error messages in the install.sh script with attempting to pull down WiringPI:

_**Would you like to install wiringPI? This is required to control the GPIO (y/n): y
Cloning into 'wiringPi'...
fatal: unable to connect to git.drogon.net:
git.drogon.net[0: 188.246.205.22]: errno=Connection refused
git.drogon.net[1: 2a03:9800:10:7b::2]: errno=Network is unreachable**_

./install.sh: line 43: cd: wiringPi: No such file or directory
./install.sh: line 44: ./build: No such file or directory
Reading package lists... Done
Building dependency tree
Reading state information... Done
python-setuptools is already the newest version (40.8.0-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
./install.sh: line 54: easy_install: command not found
Reading package lists... Done
Building dependency tree
Reading state information... Done
python-dev is already the newest version (2.7.16-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree
Reading state information... Done
libpcre3-dev is already the newest version (2:8.39-12).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Could not open requirements file: [Errno 2] No such file or directory: 'requirements.txt'

Would you like Gembird USB Support (y/n):

The repo at http://git.drogon.net/ seems to be nonexistent. You receive the information below when visiting that site.

"git.drogon.net is temporarily unavailable. Please look for alternatives for wiringPi, etc. -Gordon"

Please advise on how to proceed.

Rawlings123, 2020-05-10:
I am having the same issue. Did you find anything that works?

scottydelicious, 2020-07-10:
Can you edit line 42 in install.sh?  Change the following from
`        [Yy]* ) git clone git://git.drogon.net/wiringPi;`
to
`        [Yy]* ) git clone https://github.com/WiringPi/WiringPi.git;`

knezi, 2021-08-21:
@scottydelicious Lines 43 and 45 need to be changed accordingly too. Replace `wiringPi` with `WiringPi`.