craftbeerpi/craftbeerpi #241: Craftbeerpi wont start..

Paulianer92, 2022-01-12:
Hi!

I have installed Craftbeerpi following the instructions and it been set to autostart on boot. 
The problem is that it don't. When I try to start it manually I get no response. It's the same if I try getting a status.

Any ideas on what might be wrong?

[image]

jkissner, 2022-02-28:
anyone else having this issue?

jkissner, 2022-02-28:
did some more digging and it's busted.  the entire install is dated.

sneitsa, 2022-03-11:
I have the same issue, seems to be related with flask

sudo python runserver.py 
HALLO
/usr/local/lib/python2.7/dist-packages/flask/exthook.py:71: ExtDeprecationWarning: Importing flask.ext.babel is deprecated, use flask_babel instead.
  .format(x=modname), ExtDeprecationWarning
/usr/local/lib/python2.7/dist-packages/flask/exthook.py:71: ExtDeprecationWarning: Importing flask.ext.wtf is deprecated, use flask_wtf instead.
  .format(x=modname), ExtDeprecationWarning

Paulianer92, 2022-03-12:
I solved it by followig this guide https://openbrewing.gitbook.io/craftbeerpi4_support/readme/server-installation and installed Craftbeerpi 4 from a pre-configured image to your sd-card.